FrameworkComputer/Framework-Laptop-13 #2: Schematic

ddavidebor, 2022-04-23:
Please release schematic and Boardview to allow DIY repairs and mods

Hellreaver, 2022-04-30:
Agreed. I believe framework stated they were thinking about it, but I would greatly appreciate it - I believe it's a huge leap in the right direction. Appreciate the consideration, Framework team.

donotnoot, 2022-05-03:
I'm pretty sure that they can't do that because of NDAs and such. Or at least that's what it seems that Louis hints here after he spoke to them about this. https://www.youtube.com/watch?v=G2YjKYG8P58 

If you have a repair business, have a FW in for repair, and are happy to sign some NDAs, then you might be able to get them according to this: https://knowledgebase.frame.work/availability-of-schematics-and-boardviews-BJMZ6EAu

eclecticc, 2022-05-12:
@donotnoot thanks, that is correct.  We've uploaded what we are able to here: https://github.com/FrameworkComputer/Mainboard/blob/main/Electrical/Mainboard_Interfaces_Schematic.pdf

closed 2022-05-12 by eclecticc

bmcdonnell, 2023-08-05:
> @donotnoot thanks, that is correct. We've uploaded what we are able to here: https://github.com/FrameworkComputer/Mainboard/blob/main/Electrical/Mainboard_Interfaces_Schematic.pdf

For anyone's future reference, I see the schematic PDFs have since moved here:
https://github.com/FrameworkComputer/Framework-Laptop-13/tree/main/Mainboard