FrameworkComputer/Framework-Laptop-13 #6: Input board connector part number confirmation

CRImier, 2022-05-08:
Designing a PCB mating with the input board connector, I've noticed that motherboard pictures showing the input connector don't match the Pinouts.md-stated 10156000 connector pictures on Mouser and Amphenol website. For instance, there's large pads to the sides of the connector, which are not present on the Amphenol 10156000 connector (and not just pictures - they aren't shown in the connector drawings and recommended footprint, either).

Motherboard picture:
[image]
[Mouser](https://eu.mouser.com/ProductDetail/Amphenol-FCI/10156000-051100LF) and [Amphenol website](https://www.amphenol-cs.com/bergstak-0-50mm-self-alignment-10156000051100lf.html) picture of the connector
[image]

Insights?

CRImier, 2022-05-08:
[image]
Hmm, seems like I've been referring to an outdated motherboard picture? Lack of pictures is a bit of a problem.

eclecticc, 2022-05-12:
Yep, the image with the ground tabs was an early prototype that used a different connector.

closed 2022-05-12 by eclecticc