FrameworkComputer/Framework-Laptop-13 #7: M.2 E-key pinout

CRImier, 2022-05-08:
Board designers get quite a bit of freedom when it comes to wiring up M.2 slots - as subsets of designed-in functionality can be omitted, or pinouts can be changed, in some cases (i.e. CNVI taking up SDIO pins, as can be seen in Lenovo S340 schematics). I have questions about the wiring of the E-key (commonly used for WiFi card) slot, which parts of the E-key socket specification it implements, and any extras it might have.

How many PCIe *links* does the E-key slot have - one or two? As in, one x1 link or two x1 links? Does it have I2C? What kind of control signals does it have? Does it have SDIO or CNVI wired up? Would appreciate any extra details. Ideally, there'd be a full pinout available in `Pinouts.md`.

eclecticc, 2022-05-12:
Added here https://github.com/FrameworkComputer/Mainboard/blob/main/Electrical/Mainboard_Interfaces_Schematic.pdf

closed 2022-05-12 by eclecticc

CRImier, 2022-05-12:
Thank you so much - and I hope you enjoy what we already got in mind, having looked at these ^__^