FrameworkComputer/Framework-Laptop-13 #9: Keyboard backlight, caps lock, fingerprint RGW - any current limiting on the motherboard?

CRImier, 2022-05-11:
I'm developing breakouts for the input connector, and I realized I have no idea whether there's any current limiting on these pins. Are there any resistors (or other current limiting mechanisms) on the mainboard before any of these signals reach the input connector?

num | name | notes
-- | -- | --
27 | CAPS_P | 5V
28 | CAPS_N |  
30 | KBL_P | 5V (PWM)
32 | KBL_P | 5V (PWM)
45 | FPR_LED_R |  
46 | FPR_LED_W |  
47 | FPR_LED_G |  
48 | FPR_LED_COM | 5V

eclecticc, 2022-05-12:
CAPS and FPR_LED signals are enabled using PJT138K MOSFETs.  The KBL is switched with a AO3413.  All of the signals except the keyboard matrix are also ESD protected using AZ1045-04F parts.

closed 2022-05-12 by eclecticc

CRImier, 2022-05-13:
Thank you, the schematic provided clears it all up! For anyone who could use this info in short form - CAPS_P is created with a 1.33k resistor from +5VS, and CAPS lock LED is driven low-side. Fingerprint LED is driven low-side and has no such limiting. Backlight has no limiting, and is driven high-side, with two pins used as described in #5 .