FrameworkComputer/Framework-Laptop-13 #18: SLDPRT Files for Mainboard Case

GraffJosh, 2022-08-09:
I suppose this is a feature request of sorts, but I work in solidworks and would love to use these solid models natively. Do you have SLDPRT files available? Do you recommend an STL editor? Thanks for your work so far.

joshuakravitz, 2024-03-18:
Solidworks can natively import these file types. Are you looking for Native Solidworks files (editable?)

GraffJosh, 2024-03-18:
Yes. It helps to have the parameterized files.

joshuakravitz, 2024-03-18:
So that assumes that the parts were originally created in solidworks; the issue with native CAD files is that they are not cross-compatible or backwards compatible. So unless the files were created in the EXACT version of solidworks you were using the original files are no good.

joshuakravitz, 2024-03-18:
Let me mesh the STL files and create STEP files; this may help you in modifying the files to your liking.

joshuakravitz, 2024-03-18:
There is a file called "printable_case_full" that will get you what you need as a starting point in Solidworks

joshuakravitz, 2024-03-18:
the files were created in PTC CREO Parametric 2022; so it is highly unlikely that you will get Native Solidworks files.

GraffJosh, 2024-03-18:
Ah yes, actually I found that it did get me a good starting point. 

I used an STL I found of the Mainboard somewhere (possibly this repo, though it's been a long time since I've looked at this issue), and it has worked well enough for my needs.

This request is more of a wishlist item than an urgent need.