FrameworkComputer/Framework-Laptop-13 #22: eDP connector pin 1 location

basketofkittens, 2022-08-15:
I've been working on a project where I need to interface with the eDP connector on the Framework mainboard. I just recently got a working board spun up and when I connected it to the mainboard I was seeing +17.6VB_INVPWR hard shorted to ground. I did a little bit of poking with a multimeter and I believe the pinout there is flipped in relation to mine. 

The 20879-040E-01 (board mounted connector) datasheet does not specify which pin is 1, but the drawing for the mating COTS cable sold by Digikey (PN 81466-100B-02-D) clearly shows a pin 1 location:
[image]

I followed the convention from the 81466-100B-02-D drawing and it seems the Framework mainboard is reversed. Can someone on the Framework team confirm this? 

I don't love the idea of spending money on another expensive board spin to remedy this. Can someone recommend a vendor that would be willing to do a custom Cabline-UM harness to flip the pinout?

Thanks!

imnotjames, 2022-08-15:
I have a breakout board for the edp connector.  I'll do my best to test it with a multimeter to confirm what you're seeing.

> Can someone recommend a vendor that would be willing to do a custom Cabline-UM harness to flip the pinout?

I'd be interested in this as well, but I haven't seen anything like that.

basketofkittens, 2022-08-15:
@imnotjames thanks for checking on that!

I've emailed a few domestic micro-coax harness vendors that advertise Cabline-UM assemblies. So far one has told me they have no stock and I'm still waiting to hear back from the others. I haven't dipped my toe in with any overseas vendors yet.

eclecticc, 2022-08-15:
We've gotten small volume cable assemblies before from https://www.darloxcn.com/

I would recommend changing the board though rather than the cable, since prototype cables will be less robust and reliable than the production ones in the laptop.

Here is what the pinout looks like on the PCB:

[image]

basketofkittens, 2022-08-15:
@eclecticc thank you for the information! The mainboard pinout is definitely flipped in relation to mine. That's unfortunate. 

Thank you for the vendor suggestion. Your point about production cables is well taken, but a little less relevant in my case seeing as I'm not using the Framework display or cable. I've ended up with an identical Cabline-UM connector on my display board since the above COTS cable was the easiest one for me to source at the time. That being said, I'm still leaning toward re-spinning the board since others wouldn't need to get a custom cable if they decide to use my design for their own purposes.

Would it be possible to add a note to the pinouts.md file in this repo so as to help others avoid the same mistake in the future?

imnotjames, 2022-08-16:
Oof, after testing my breakout this explains a lot of why I was having trouble.

I can confirm this is "backwards" from expectation based on the data sheet and as I understood other pin outs

eclecticc, 2022-08-16:
I added a note in the pinouts file for clarity.  Thanks for reporting this.

closed 2022-08-16 by eclecticc