FrameworkComputer/Framework-Laptop-13 #33: Keyboard Font, dimensions, 3d models

tekchip, 2023-12-02:
In attempting to get the clear keyboard etched I've discovered a lack of information regarding the keyboard.

It would be highly useful to myself, and the rest of the maker/DIY community to have the font, and exact variation used (Medium 500?), any other legends for non-text keys, the exact dimensions for the keyboard keys, and space between keys etc. Bonus points for 3D models of each different size of key. Please and thank you!

DeflateAwning, 2024-08-18:
In case anyone's looking for it, there are _some_ models in [Captain-Pumpkinhead's fork](https://github.com/Captain-Pumpkinhead/Framework-13-Vanilla-Reconstruction)

DeflateAwning, 2024-08-18:
Update: the `Framework Laptop 13 CAD.stp` in the repo now has quite high-quality keyboard models.

[image]

eaglesemanation, 2025-04-16:
Exported keys from step file into SVG using FreeCAD TechDraw workbench.
Found similar enough looking font called Futura Cyrillic Medium.
Combined in Inkscape. Here is the result:

[image]