FrameworkComputer/Framework-Laptop-13 #35: Error in Mainboard_Interfaces_Schematic_7040_Series.pdf block diagram

regaly, 2024-02-25:
There are double PCIe link(x1 Lane 15) between P52 M.2 wlan/bt slot and APU on page 1. I think one should be USB 2.0, but you've forget to edit after copy-paste.

WizardTim, 2024-04-09:
Agreed, page 4 does show USB 2.0 (port 6) and a single PCIe x1 (lane 15) on the connector which is correct for E key M.2.

However PCIe 2.0 also doesn't make much sense as lane 15 on Phoenix is a PCIe 4.0 interface. This would also bandwidth limit WiFi 7 upgrades like the Intel BE200 so I assume that's a typo as well? Although I wouldn't be surprised if it's only qualified for 3.0 speeds due to routing.

KosmX, 2024-08-19:
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
  iSerial                 1 0000:c1:00.3
```
and BT is on this USB not on PCIe
```
Bus 001 Device 005: ID 0e8d:e616 MediaTek Inc. Wireless_Device
```