FrameworkComputer/Framework-Laptop-13 #42: Camera module 2

phodina, 2024-09-08:
Hi @eclecticc ,

firstly thanks for the updated camera module!

Are they pin to pin compatible with the original camera module that you describe here? https://github.com/FrameworkComputer/Framework-Laptop-13/tree/main/Webcam

Since the OV08X camera sensor supports just MIPI interface how do you connect it to the AMD CPU? Or is there some co-processor along the way that does the Image processing DSP operations (3A algorithms)? Or just some MIPI to USB bridge that pushes the RAW image to AMD processor that then leverages GPU to process the image?

Also where can I get the datasheet for the sensor to get information about the registers and modes of operations? 

What's the best way to modify the sensor registers over I2C? `i2c-tools` or `libcamera` (or something else)?

amstan, 2024-09-08:
The connector to the webcam is compatible, yes. The only way to talk to the camera (indirectly) is through usb (that's the only interesting data pins going to the camera), so you're probably talking to some kind mipi to usb bridge (i don't have the module schematics in front of me right now tho). Based on those facts, at first glance, you're not likely to be able to get access to the sensor registers easily.

Here's an lsusb dump which should match what you would get (+- some firmware version differences):
[webcam.lsusb.txt](https://github.com/user-attachments/files/16921144/webcam.lsusb.txt)

JohnAZoidberg, 2024-09-11:
Like @mstan said, the DSP uses the USB UVC protocol. No MIPI access from the host.
So you can use UVC driver and tools to acess the camera.

On Linux it's handled by the v4l2 kernel driver. You can use the [v4l-utils](https://linuxtv.org/wiki/index.php/V4l-utils) to interact with the camera through USB.

Is there anything specific that you're interested in?

phodina, 2024-09-12:
@JohnAZoidberg I work on products that run on embedded Linux and interface with cameras over MIPI and also use Framework 13 AMD laptop so I'm curious how you handle the image pipepline.

Also I got recently the  new webcam module and would like to do some comparison on them.

Also I own Lenovo ThnikPad X13s Gen1 with Snapdragon SoC wehre the MIPI lines are directly attached to the processor. The only issue is that on Linux the IDSP does not work and the RAW image has to be processed by the CPU :-/

phodina, 2024-09-13:
@JohnAZoidberg do you also use the RTS5803 MIPI to USB brigdge?

Do you have datasheet for the IC?

JohnAZoidberg, 2024-09-14:
We've got the DSP names on our marketplace.
https://frame.work/tw/en/products/webcam-module

RTS5879 for 2nd gen camera.
RTS5853 for original camera.

JohnAZoidberg, 2024-09-14:
Since the modules are standalone and can be used on lots of different of our mainboards. Even across 13 and 16 inch, all tuning and calibration data is stored in the module.
Tuning is done together with the framework laptop bezels.

Sorry Realtek does not make the data sheets public right now.

phodina, 2024-09-15:
Understood and thanks for the part reference and info on the calibration data.

phodina, 2024-09-16:
Would it be possible to specify the max amount of current you support on the 3.3V rail?

Do you use the ambient sensor to adjust the camera or is that just for the screen brightness?

JohnAZoidberg, 2024-10-14:
> Would it be possible to specify the max amount of current you support on the 3.3V rail?

How much current the module would draw? I'm not sure. Why are you asking?

> Do you use the ambient sensor to adjust the camera or is that just for the screen brightness?

Neither of those is done in firmware. Windows and Linux both adjust screen brightness based on ALS sensor readings.
Neither of them adjust anything related to the camera, as far as I'm aware.

phodina, 2024-10-21:
> How much current the module would draw? I'm not sure. Why are you asking?

Well, since I replaced my webcam module I have the old one which could be used in some project. So that would help to estimate the power supply.