FrameworkComputer/Framework-Laptop-13 #46: Document closed-case debugging in schematic

WesleyAC, 2024-12-30:
[This blog post](https://www.howett.net/posts/2023-04-framework-ec-card/) details how the CC pins on the back right USB port can be used to convert the sideband pins into a UART that talks to the EC. However, this is not reflected at all on the schematic, as far as I can see. I would love to see this shown, either in the block diagram or in more detail on a later page.

amstan, 2024-12-30:
Yeah, it's on my list to do this for both closed case and open case debugging. Let me know if you have any questions in the meantime.

Btw: there's also https://i2clabs.com.au/ec-debug-expansion-card/

WesleyAC, 2024-12-30:
Thank you! And yeah, I purchased one of those boards last night :)

I have a couple related questions:

* Are there other testpoints associated with the NPCX993 chip? For instance, are JTAG testpoints exposed?
  * In general, I would love a complete list of testpoints on the mainboard somewhere
* In general, are there good places to look for open documentation about the NPCX993? The only thing from the manufacturer I can find is [this marketing material](https://www.nuvoton.com/export/sites/nuvoton/files/product-related-information/NuvotonPSG.pdf) which talks about the capabilities. [The chromium EC source](https://chromium.googlesource.com/chromiumos/platform/ec/+/master/chip/npcx/system.c) sort of documents some stuff, is there any better place? Alternatively, is there any reasonable way for a individual to get access to the datasheet (even under NDA?)