FrameworkComputer/Framework-Laptop-13 #47: Request for keyboard wiring diagram.

Ben-PH, 2025-01-10:
I have an issue with my touch pad ribbon cable being damaged, with only some of the keys working.

I've since resolved my issue (need to get a replacement), but in retrospect, being able to cross reference functional/non-functional keys to which parts of the matrix is/isn't working would have been nice.

I'm sure there are other use cases as well.

eclecticc, 2025-01-12:
It's actually documented in the ec code, and dhowett has a nice
visualization here: https://www.howett.net/data/framework_matrix

JohnAZoidberg, 2025-10-23:
fyi @alex-courtis

alex-courtis, 2025-10-23:
Cheers @JohnAZoidberg !

With that and the scancodes I can `--remap-key` my keyboard as dvorak.

alex-courtis, 2025-10-23:
> If you have the full dvorak mapping, feel free to post it here or open a PR to https://github.com/FrameworkComputer/framework-system We could add a command to do the full dvorak mapping in a single run

I'd be really glad to do that! I can do colemak as well.

Looking at the matrix above I'm not sure which one is the framework logo key - is it 2 14?

JohnAZoidberg, 2025-10-23:
> I'd be really glad to do that! I can do colemak as well.

I started doing this to remap multiple keys at the same time, to emulate a Mac layout
https://github.com/FrameworkComputer/framework-system/commits/mac-layout/

alex-courtis, 2025-10-23:
> I started doing this to remap multiple keys at the same time, to emulate a Mac layout https://github.com/FrameworkComputer/framework-system/commits/mac-layout/

Very nice - that will be a great convenience for users.

I am still struggling with the scan codes and can't find a map of their values. I've looked at the linux [include/uapi/linux/input-event-codes.h](https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h) however they don't correspond with the scancodes in the example e.g. left ctrl value `0x0014` maps to `KEY_T` which is clearly incorrect.

I'd be really grateful if you could show me the "source of truth" for the scancode values.