FrameworkComputer/Framework-Laptop-13 #48: Consider enabling org-level discussions on GitHub

spenserblack, 2025-01-12:
Apologies for raising this issue here, but I couldn't find a suitable repo to make this request.

I think it would be nice to enable organization-level discussions ([docs](https://docs.github.com/en/discussions/quickstart#enabling-github-discussions-on-your-organization)). This would help GitHub users discuss Framework laptops in a general location, which might not be related to a single repository.

This request is partially resolved by the existing community.frame.work site, but org-level GitHub discussions can:
* Allow users with existing GitHub accounts to discuss with each other
* Allow for a space that, while not specific to an individual Framework repository, is specific to GitHub users who are interested in Framework

eclecticc, 2025-01-12:
I definitely understand the ask, but as a small team we are very constrained on bandwidth to monitor and moderate multiple communities.  Even with our Community Forum and Discord, we are stretched thin.  We'd like to keep general discussions centralized to the Community Forum.

closed 2025-01-12 by eclecticc

spenserblack, 2025-01-13:
That makes sense.

Anyway, I was hoping to have that central location so that I could ask if you'd be willing to set your organization up for GitHub Sponsorships. I really love your product and mission, and was hoping to provide just a little bit of extra support 🙂