FrameworkComputer/Framework-Laptop-13 #55: Camera privacy screen

phodina, 2025-05-03:
Hi @JohnAZoidberg ,

is there a way to change the privacy screen if it's active?

I guess it's an image that's hardcoded to the FW in the webcam controller as it's being received in the UVC stream over USB and is not part of the kernel nor userspace, right?

[image]

JohnAZoidberg, 2025-05-03:
Yeah sorry, no way to change it. It's embedded in the camera firmware.

phodina, 2025-05-03:
Okay, so the only way would be to detect it in stream and replace it with pipewire, right?

Btw is there a way to update the firmware on the controller? Like force it to DFU or something like that?

JohnAZoidberg, 2025-05-06:
> Okay, so the only way would be to detect it in stream and replace it with pipewire, right?

No actually there's a UVC bit that you can look at to detect whether the switch is flipped or now.
See: https://github.com/soyersoyer/cameractrls/pull/51
You could hook pipewire to detect that and replace the image.

> Btw is there a way to update the firmware on the controller? Like force it to DFU or something like that?

The firmware can be updated using DFU and we distribute firmware updates on fwupd.
But unfortunately the firmware does not support updating the mute image using DFU 😓 
That needs to be done using a proprietary extension.

phodina, 2025-05-11:
> No actually there's a UVC bit that you can look at to detect whether the switch is flipped or now.

Awesome, that's a good way to swap the image.

> The firmware can be updated using DFU and we distribute firmware updates on fwupd.

Nice that it supports the DFU and it's integrated into the LVFS. Thanks for the effort!

> That needs to be done using a proprietary extension.

Do you mean it would require also some hardware tools or just some extensions on top of the DFU( or some other protocol)?

Any chance to release some open source tools? Or are there let's say tools for Windows to capture the USB traffic and reverse-engineer that?